openLRSng/openLRSng #198: Is there a discussion 'area' or are we posting questions and stuff in the "issues" tab on Github

Rusty105, 2018-02-27:
Is there a place for discussion, or is all discussion happening here in the Issues tab on Github?

mikeller, 2018-02-28:
Opening an issue with a proposal / discussion item / feature request, and then having a disussion on it is how it is done in openLRSng. This has the advantage of creating a permanent record of any insights and conclusions that are reached.

Rusty105, 2018-02-28:
OK, wasn't sure, and wasn't sure how often anyone is checking for new posts

ameeuw, 2018-02-28:
Even though it might be good to sometimes take information that is buried within issues and transfer them to the wiki. For example the information on how to enable multiprotocol (and exactly how to add the resistor between ppm and rx pins) is rather hard to find..