openLRSng/openLRSng #199: Refrence design for existing hardware.

Rusty105, 2018-02-28:
Ok, trying to help out here...

There are several suppliers of OpenLRSng hardware; Hawkeye/DTH UHF, Hobbyking, Flytron, Brotronics, etc... Is there an existing "reference design" regarding hardware? Such as What pins are used for what purpose on the microcontroller? There has to be, right? Otherwise the code would not work. And is there a ref design for what types of outputs are available on each channel? Has to be.... right?

The reason I ask is that if we are going to move forward I think we need some sort of road map. Something for new hardware manufacturers to follow, something for the DIY crowd to follow I guess I could go through all the schematics available and build a reference, but have no way of knowing if that is the approved reference. The same goes for new proposed designs with updated RF modules and microcontrollers, but that is another topic to open....

ameeuw, 2018-03-01:
Not sure whether this helps, but I think that posts [#69](https://www.rcgroups.com/forums/showpost.php?p=28252876&postcount=69) and [#70](https://www.rcgroups.com/forums/showpost.php?p=28252876&postcount=70) by felipealex in the rcgroups thread define kind of a reference design for TX and RX for the SI and RFM modules.

ameeuw, 2018-03-01:
For moving forward I think it would be necessary to utilise a processor that is able to output serial channel data and take serial telemetry from a flight controller back to the radio.. which is (to best of my knowledge) not possible with the ATmega328.

geofrancis, 2018-03-01:
just a thought, what about a atmega2560? its got the uarts and it should be very simple to port from a 328?

https://www.ebay.co.uk/itm/Mega2560-CORE-smart-mini-2560-ATMega2560-Arduino-Compatible/221502571388?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m1438.l2649

mikeller, 2018-03-01:
@ameeuw: It is possible in a limited form: If the RC protocol only uses communication from the RX, and the telemetry protocol only uses communication to the TX, then both can share the TX / RX pin on the openLRSng RX module. (I am using this configuration with Betaflight, with SUMD and FrSky hub as the protocols.)

Using a MCU that supports more than one UART would definitely allow a wider choice of protocols. But it would probably have to be replicated in the TX module as well, to support reception of RC control data and bidirectional telemetry at the same time.

Rusty105, 2018-03-01:
OK, went over the code a little more, specifically the 'hardware.h' file and found the pin assignments for the various "hardware types" Such as this:

> #define PPM_IN           A5 
> #define RF_OUT_INDICATOR A4 
> #define BUZZER_ACT       9 
> #define BTN              10 
> #define PPM_Pin_Interrupt_Setup  PCMSK1 = 0x20;PCICR|=(1<<PCIE1); 
> #define PPM_Signal_Interrupt PCINT1_vect 
> #define PPM_Signal_Edge_Check ((PINC & 0x20)==0x20) 
> 

So what I think I am really looking for is a list of the definitions such as "PPM_IN" and "BUZZER_ACT" that is used in the code.  I think if hardware developers knew what options were currently implemented in the code, and knew they would stay there in the future, they may be more willing to develop newer hardware.  Or in the case of moving to newer chips, maybe a design standard that hardware developers could use. NOT saying it has to be a strict standard, as not everyone will want all options, but for a core number of things, like "PPM_IN", the ppin assignment should probably stay the same...... ?

artemen, 2018-03-06:
Sorry guys I'll chime in with my 5c ... 
Out of my experience with openlrs hardware (about 3years now) all of the openLRS RX designs badly need filtering of power source just before the RFM22/23 here is a video of how the offthe shelf hawkeye 9ch RX, delux v1 TX and my own version of 1W TX affected by noise on power supply. The last one shows noise floor below -114db while delux v1 at -100db and 9ch Rx at about -108db. In the video there is a short drop in db of delux v1 when I accidentally touched antenna with my hand. In all cases power supply is my PC's USB port. https://photos.app.goo.gl/X3Y9YHd3bG67J29x1

atomiclama, 2018-03-06:
@artemen Can you give some advice on the filtering required? Are you doing anything special or just decoupling caps? Any tips?
I'm knocking up my own HW with just a selection of caps on the supply line, 1uF 100nF 10nF 
Cheers

artemen, 2018-03-06:
I use 100uh 1A rated coil right before the power pad of the rfm23bp (in this case).

marcdornan, 2018-03-06:
@artemen I was looking at the Dolphin design schematic you did for ULRS (I have a few that I am converting to OpenLRS) and I do not see this. Assuming it is not there do you think I can add a ferrite bead RF attenuator to before RFM Vcc pad? Or would a coil be better. I picked up a load of these a while back. https://www.ebay.com/itm/ACT-Ferrite-Shield-Bead-EMI-RFI-Suppressor-200-LOT-/172596471645?hash=item282f8d275d

artemen, 2018-03-06:
its the very first schematics I believe, later on, I've shared this board: 
https://oshpark.com/shared_projects/nzfukPxB

marcdornan, 2018-03-06:
Yes — I have the first version of that board. (The USB needed be shaved off
when the antenna was attached).

I did not see a coil inductor on the schematics for the RFM. It is not in
front of me now. Am I wrong? Should I add one?

artemen, 2018-03-06:
I do not believe it is possible to add one to v1, just use the v1.5 board (its only about $4 shipped within the US), the only change I believe was that inductor and rearrangement of ports, I've mentioned that somewhere on the rcg thread though... (i believe :) )  I didn't update the schematics, I use this one: SRR0735A-101M

marcdornan, 2018-03-06:
Got. Mine are already made up and working. So I think I do want to add one.

I am powering the output stage of the RFM at a higher voltage (7v) than the
VCC input (3.3v) so I get close 2W. So it is easy as I have basically
already got an external 3.3v regulator (also added an extra RF transistor
in the empty pads and changed a resistor value). Just need to add an
inductor. I think that it is the 3.3v VCC of the RFM that needs inline
filter.

Well it cannot do any hard to add an inductor.

marcdornan, 2018-03-06:
> I must say, that I've seen what narpat is doing, its very crude and
> potentially dirty, especially messing with the rf matching caps, though 7v
> is OK, it shouldn't give more than 1500mw, in any case, rf switch on
> rfm23bp (UPG2179 in SOT363) has abs max rating of +33dBm (1995mw). running
> anything at abs max is not a good idea in long term and certainly not above
> that. the biggest concern is that rfm23bp were rf balanced and filtered
> given the board topology for a given max output +30dBm, to check how this
> really affects its performance we need an anechoic chamber. what's the
> point of gaining 2dBm if you'll be swamping your 1.2ghz video with
> harmonics?
> in any case, both 3.3v (powers SI4430) and 7v (powers the amp mosfet) need
> to be very clean.

Probably right. Some of his mods are extreme. I do not touch the caps and
7v was all I thought prudent.

My mods have fixed the heat issues with one resistor that was running way
out of spec--screechingly hot at high duty cycles. Also the extra rf
transistor is really needed I think if you are running at anything over
3.3v.

Running the module at 3.3v seems to be beneficial. Whether it makes any
sense to go as high at 7v on the output I do not know yet! It works well
and my GPS is ok.

Rusty105, 2018-03-06:
Hey guys...... Any links to theses mods?

marcdornan, 2018-03-06:
You will have to trawl the ULRS thread. I got so fed up with the OP that I
threw my stuff in a box. Just taken it out again. I think what I did is
sound but it has to be validated by a long range flight. Take note of
@artemen's cautions. Personally I would not run the rfm23b at anything over
5v without some of these mods.

Rusty105, 2018-03-06:
Yeah, been lurking in that thread too, can't believe the stunts that guy pulls (the OP)

artemen, 2018-03-06:
@marcdornan 
hey, if you are modding one of the dolphins, the best way is to actually to set lm317 to output 7-8v, than grub 5v from mcp1700 and feed it to XC6206 which is onboard of rfm module. this way you don't need a separate vreg as lm317 on dolphin is properly implemented and has around 66db PSRR and 0.01%V typical line regulation.  In this case, cut only the trace connecting XC6206 to rfm's VCC pin, than move the decoupling caps right on the XC6206 Vin pin (use a bit of creativity there). 

mcp1700 outputs clean 5v at 250mAh, atmega uses about 20-30mAh si4330 max 86mAh, so about only half of capacity. 

LM317 can provide 1.5A without breaking a sweat, so if at 6v we were drawing about 600mAh at 7-8v 1.5A capacity is enough.

marcdornan, 2018-03-07:
Actually that is pretty much what I am doing. I spent a bit of time
checking the rating of the various vregs already on the Dolphin and
realized I did not need the extra 3.3v reg and did exactly what you
suggested. It is working well thus far. We will see how far I fly my Mini
Talon with this set up.