openLRSng/openLRSng #202: Hopinterval / Hoptimeout

thedillybar, 2018-03-17:
Has anyone debugged a connection to see how accurate the Hopinterval calculation is? And looked at a connection with lost packets to see whether or not it stays in sync?

My concern is that if a few lost packets causes it to lose sync, a small problem becomes a larger one. We could recalculate the Hopinterval (and timeouts) based on successfully received packets. 

I also noticed that the TX doesn't hop once it loses its input PPM signal (ppmAge >7). It could continue hopping each interval even though it's no longer sending packets. Maybe not worth it because it's going to lose sync anyway?

artemen, 2018-03-19:
you might be onto something, this is my post from pwm amp thread:
I've recently decided to do the [Narpat's](https://www.rcgroups.com/forums/showpost.php?p=33899703&postcount=3691) mod and power PA separately from the Si4331, as expected at 9v we do get around 2w of output from the old rfm23bp, also if done properly heat is not excessive and PA runs for more than 24hrs without frying. this was demonstrated by Narpat a year ago. What's not expected is that even with about 33dbm output, packet loss rate didn't change much from stock dtfuhf delux v1 running at 29dbm, this made me think that either RX or TX or both do not get locked properly onto the exact frequency. If so, would this be hardware fault or code fault?

maybe we are not precise enough calculating sync? if its atmel limitation, there used to be a 32bit fork...   regardless, creating a simple 32bit test hardware should be relatively easy, chip and fast.

marcdornan, 2018-03-19:
Glad to see you have verified that mod. I always felt it was good — I have
only just decided to buy some RF test gear as I am tired of not just not
knowing if a mod is successful.

thedillybar, 2018-03-20:
If it were a frequency issue, you would see packet loss even with the TX and RX very close together. 

The first question is how often do they re-sync? I assume at 0% packet loss, they never re-sync (although worth testing). But what about 10% or 50%? When they do re-sync, were they actually out of sync or did we trigger a re-sync that wasn't needed?

The goal should be zero re-syncs. Some bands require minimum 50 hopping channels. These systems must be robust against re-syncs because they could take a long time. 

How to go about debugging this? Is it straightforward to connect the RX or TX to openlog via serial? Or to output it in the PPM stream and monitor that from a computer?

artemen, 2018-03-20:
Its more like center frequency is missed, afaik hop channel width is about 5khz?!? if i am not mistaken, so if center is 2khz off tx and rx will still be fine close by at 30db probably at lest 1km+ LOS,  as they still have 2-3 khz overlap. This would also explain sudden drop of rssi even at 200-300meters....

WidiTaufik, 2018-03-20:
Hello, artemen it seem that Si4432 modul able to cope with frequncy offset well, atleast what the datasheet said. IMO the packet lost is about hop/sync issue
[image]

artemen, 2018-03-21:
@WidiTaufik 
than, maybe it AFC settings are too wide and we get quite a lot of noise in the mix?

WidiTaufik, 2018-03-21:
@artemen I'm not sure about that
I think packet lost is not really a problem if you still can get 50 packets per second, so you will have servo refresh rate of 50Hz

artemen, 2018-03-21:
we really need only about 30 packets per second (human's reaction limitation) so if RX can just continue outputting last received packet until it gets the next one we should be fine as long as there are at least 30 good packets each second.

artemen, 2018-03-21:
what are the conditions under which resync is triggered?

thedillybar, 2018-03-22:
`hopTimeoutSlow = hopInterval * hopcount;`

`  } else if ((numberOfLostPackets == hopcount) && (pktTimeDelta > hopTimeoutSlow)) {
    // hop slowly to allow re-sync with TX`

Once the RX misses hopcount consecutive packets (i.e. it misses the packet on every channel), then it re-syncs (hops very slowly to reestablish sync with TX).

artemen, 2018-03-22:
Well, this looks like good logic.... 
However, if neither TX nor RX were powered down they should be on the same channel even though RX might've lost all packets in a row can we still remain hopping  for at least a second before trying to resync? 

Another thought: do we really need more than 30-40packets per second? I know telemetry demands much faster rate, but I think we should emulate needed baud rate on the ground side instead of trying to squeeze all of the data. Some of it could be  discarded, I mean, we are not trying to log IMU performance trough RC.

thedillybar, 2018-03-22:
They should remain sync'd as long as neither is powered down. The getInterval function calculates how long each transmission should take based on # of bytes being sent, rate (bits/second), and whether or not diversity is enabled.

On the TX side, once the hopInterval has expired, it sends data and hops to the next channel.
```
  if ((time - lastSent) >= getInterval(&bind_data)) {
    lastSent = time;
...
      // Send the data over RF
      setHopChannel(RF_channel);
      tx_packet_async(tx_buf, getPacketSize(&bind_data));
      //Hop to the next frequency
      RF_channel++;
```

On the RX side, after receiving a packet it hops to the next channel and listens. If it doesn't receive a packet after (hopInterval+1000 microseconds), it hops to the next channel. Then it listens for hopInterval and again will hop to the next channel if nothing received.
```
void loop(void)
{
  timeUs = micros();
  pktTimeDelta = (timeUs - lastPacketTimeUs);  
...
  if ((numberOfLostPackets < hopcount) && (pktTimeDelta > hopTimeout)) {
    // we lost a packet, so hop to next channel
    willhop = 1;
...
    lastPacketTimeUs += hopInterval;
```

Yes it's easy to change the criteria for re-sync, but we need to debug first to see if that's even a problem. The getInterval function estimates an interval of 39.3 milliseconds for 8 bytes @ 4800bps (diversity disabled). If this interval were too short, the RX could hop too soon. Once a single packet is missed, the RX is 1000 us (1 ms) behind the TX. I'm not sure why 1000 us was selected and if this slight out-of-sync makes any difference.

I would prefer to leave the rate for another discussion, but assuming 8 bytes @ 4800bps (diversity disabled) = 39.3 milliseconds = 25 packets per second. I plan to use @ 4800bps with telemetry disabled.

Still looking for advice on how to debug...

thedillybar, 2018-03-22:
Apologies I have no experience with hardware debugging (except for a tiny bit with a Pickit 2 and microcontroller -- not sure if that counts). It seems there are some simple things to log that won't disturb timing.

1. How many re-syncs are triggered on the RX? (At this point it's out of sync anyway, so logging will change the timing but won't matter much)
2. How long does the transmission take on the TX, i.e. is the getInterval function accurate/conservative enough? (After transmitting, it hops channels and waits until hopInterval expires before doing anything -- there is down time here we could log)

I do have an Omnibus F4 on the way which is STM32 based & has blackbox/microSD. Not sure how to monitor the connections you mention.

artemen, 2018-03-22:
well, all of the servo outputs are also inputs, connect miso, mosi, sck, nirq  to four of those. write a sketch to read those digitally as fast as you can and write values to an SD card. later on analyse for packets, how precise is the interval between them.

mikeller, 2018-03-27:
Hey good to see you back, @csurf, I was wondering where you'd gone off to.

atomiclama, 2018-03-28:
Hey @csurf wondered what had happen to you, things went a bit quiet. Care to share what you have been upto where you hang out. Groups / repo / etc. Maybe can help out if it is not Atmega based :-)

csurf, 2018-03-28:
My plan is to publish a completely stripped-down & customized version of OLRSng, but without having any restrictions of having to maintain backwards compatibility and coding conventions that might be tied to the original repo. This will give me the freedom to rework things freely and be able to slim down some of the overhead and/or rework some of the things that might be considered somewhat clunky. I figured that this was the best route, instead of trying to completely rework and mangle the existing codebase, at the risk of possibly angering the existing user base.

If anyone is interested, I can post more details in the future as soon as I've got something that I'm comfortable publishing...

atomiclama, 2018-03-28:
Yeah, I'm interested. Keep us posted.

mikerc2017, 2018-03-28:
@csurf 
yeah thank you for your work . 
So your version have support a Smartaudio or other protocol telemetry ?

mikeller, 2018-03-28:
All the best with that, @csurf! Thanks for the work you've done in cleaning up the existing repository, and sorry if I had not made it sufficiently clear from the beginning that one of my intentions for this repository was to keep it backwards compatible and keep the owners of existing hardware supported.

artemen, 2018-03-29:
@csurf
I think we are missing packets when we really shouldn't be. And the more I watch my dvr footage the more I notice that packet loss is not consistent and irregular. For example I've just seen me flying 6.5km with 98qual at 37-45meters, than I go up to 94meters and at 5.9km I get more than a second of total packet loss so my failsafe kicks in (have it set at 1000ms delay).  I know I am flying in a very busy UHF location between two airports and and above many farms who regularly use walkie-talkies, but still I have 24hop channels for this purpose, no way all 24hops were missed for a full second at 9600baud... when clearly RF signal strength is not an issue. 
I know we've been looking at RX code mostly, but maybe there is an issue with TX packet process?

artemen, 2018-03-29:
@csurf

I gues this is the correct route to take as long as the existing hardware is supported  unless you are talking about 32bit or different RF module support, that would be a totally different and understandable thing.

thedillybar, 2018-03-29:
@csurf
Thanks for the information. I was not going to attempt shortening (optimizing) the interval -- my concern was is the current interval long enough to prevent unnecessary re-syncs. It sounds like the answer is yes. The next question is, the timeout is hopInterval + 1 ms. That means if it misses a packet on one channel, it shows up to the next channel 1 ms late. Does that prevent it from decoding the next channel as well?

Thanks again for stopping by. It will be a while (if ever) before I'm equipped with the proper hardware setup to do similar debugging.

mlyle, 2018-03-29:
N.b. things are much more robust with telemetry, because then there is additional time for the receiver to listen on each channel if the TX is "late".  That is, by reducing TX duty cycle, the need for accurate timing is also significantly reduced.

csurf, 2018-03-29:
@thedillybar 
well, that's not exactly how the link management algorithm works (assuming I understood your description correctly). 

After the initial link is acquired, on each loop iteration, an attempt is made to handle a received packet based on the `RF_Mode` interrupt flag. If a valid packet is received, the hop channel is then updated to the next channel in the sequence.

The `hopInterval + 1000` value you're referring to is the `hopTimeout` variable, which is only used to determine whether or not a packet is overdue on a given hop channel. If so, the packet is marked as lost & we force the system to hop to the next hop channel in the sequence. 

I believe the 1000 microsecond offset value is chosen because it keeps the system within a safe margin of timing error wherein there's still enough time to hop to the next channel without losing sync with the hop timing cycle. I * believe * this is where the `BYTES_AT_BAUD_TO_USEC()+ 2000` factor comes into play within the `getInterval()` function (common.h). Although I haven't verified it, I believe this is why the extra 2000 microseconds are tacked on to the interval in order to act as a buffer / margin of error so that there's still time to deal with lost packets.

FYI, this type of 'voodoo math / black magic numbers & formulas' was the reason why I wanted to get a testing & evaluation effort started in order to go back & re-evaluate each part of the algorithm; I felt it would worthwhile for us to have a clear understanding of how the system timing worked.

mlyle, 2018-03-29:
@csurf -- that's not quite how it works.  hoptimeout comes from hopinterval and already includes the 2000.  But...

The receiver takes time to settle and land on a channel.  Times are measured from completion of last packet.  If hoptimeout was exactly the interval that the TX was sending at, if the next packet was even a microsecond late because of clock skew we'd miss it.  So we wait around a millisecond longer.

The key thing is, it all comes down to the time during which the TX is not transmitting.  We must leave a channel after the TX has definitely completed sending a frame.  We must arrive on the next channel and settle on frequency before the TX begins sending a frame.  This is a very tight window without telemetry.

With telemetry, things are much better-- there's a whole bunch of time the TX is listening during which it's safe to leave a channel.

So, to summarize:  + 1000 provides a budget for reception time jitter / clock misalignment.  But it spends precious time that the TX is waiting for its tx timeslot, too