openLRSng/openLRSng #203: Broken header check

mlyle, 2018-03-29:
Previously the LSB of the header magic was XOR'd by the hop channel index.  This logic has been lost.

It's an important piece of logic, because otherwise receiver overload could allow you to receive something from the wrong channel, and think you're in sync when you're not.  If two channels are close, this could keep you permanently out of step / from resyncing.

It also means the current version of this code base does not interoperate with past OpenLRSng.

@kh4 FYI

mikeller, 2018-03-29:
Thanks @mlyle. Any more pointers to a pull request, or code location, where this happened.

tracernz, 2018-03-29:
Wow, that diff is basically unreadable. Pretty hard to expect reviewers to pick up bugs when you move code in the same commit as functional changes.

kh4, 2018-03-29:
Seems I need to look on this more closely ;)

But yeah the xorring of the channel number to magic is essential and it was added to solve the issue mlyle described.

mlyle, 2018-03-29:
Well, if you break interop, you break the STM32F1 version I am developing (which has much cleaner code and greater capabilities)...

csurf, 2018-03-29:
I don't personally agree it's an issue, but opinions vary. Anyway problem fixed, merging & closing.

mikerc2017, 2018-04-02:
Kh4 work  back    ???  happy for this