openLRSng/openLRSng #206: Cant setup serial bridge, any specific setup needed ?

tech0x, 2018-04-02:
Hello,

I use master version of OpenRLSng on the tx and rx,
any specific options needed for setup transparent bridge ?

csurf, 2018-04-04:
@tech0x 
there's probably a bug in the serial link code, as I also had issues getting it to work bi-directional with the current code base; It would work in one direction only, although ATM I can't remember if it was from RX to TX or TX to RX. 

I'll need to take a look at it, but it might be a while 'cuz I'm working on other stuff.

tech0x, 2018-04-04:
What can I do for You, to help solve this issue ?

csurf, 2018-04-04:
@tech0x 
- make sure you have 'TX' and 'RX' setup on outputs for RX module (do this via Configurator)
- make sure to setup the correct serial baudrate on the TX side (via Configurator)
- setup serial connection to both TX and RX simultaneously (using FTDI adapter on RX or something)
- use a terminal app to open serial connections to both TX and RX, using correct baudrate setting that you setup above
- test sending serial data on each terminal using keyboard
- you should see the characters appear on the other terminal (i.e., from TX to RX and from RX to TX)
- report back and let me know what you find...

If I remember correctly, it only worked in one direction...

tech0x, 2018-04-06:
Ok thank You, I will test it.
I found missed TXD option in the Channel Output, is this ok ?

csurf, 2018-04-06:
You need to select TXD and RXD in the channel outputs for the RX.

csurf, 2018-04-06:
@tech0x 
what versions of configurator & firmware are you using?
This is a bug, but it should have been fixed.

tech0x, 2018-04-07:
Both firmware and configurator downloaded from this url master version week or more time ago.
https://github.com/openLRSng/openLRSng

But I test version 3.8.8 and they works fine.
Ok I download current master and test them again.

tech0x, 2018-04-07:
I make the test.
Today current master branch have this problem.

csurf, 2018-04-07:
@tech0x 
ok, I'm working on it.
what type of TX and RX hardware are you using?

csurf, 2018-04-07:
@tech0x 
there's a pending PR that should fix the Configurator issue, but it won't necessarily resolve the serial bridge issue. If & when the PR gets approved, you'll be able to re-pull master and test again. 

Unfortunately, I don't have time to fully test the serial bridge behavior right now. You'll have to wait until the PR is approved and merged so that you can try it out. Or, if you know how to pull a PR, you might want to try testing the PR directly. Then, we can proceed with testing&debugging the serial bridge issue...

csurf, 2018-04-07:
@tech0x 
please try testing from master now...

csurf, 2018-04-08:
@tech0x 
ok, so we've confirmed that TX->RX serial bridging works, but it's broken in the other direction. I'm pretty sure where in code the problem exists. I'll take a look at it in a while.

csurf, 2018-04-08:
update with some test results:
- 3.8.8 TX with 3.9.0 RX, serial bridge works fine in both directions
- 3.9.0 TX with 3.8.8 RX, connection / link loss issues, can't get both to maintain link sync'ed. FYI, this relates to #205 

There's definitely an issue on the TX side. 
Will have to look into this further.

csurf, 2018-04-09:
@tech0x 
I found & fixed the problem.
try downloading & testing from master again as soon as PR #210 gets merged...

tech0x, 2018-04-09:
Thank You,

at first look all works,
but at start terminal session rx get few unknown symbols
 ��test
321

and tx 
OpenLRSng TX starting 3.9 on HW 5
  �test
123

tech0x, 2018-04-09:
Last RX version  works on Atmega 168, great works guys!

PR #209 (2018-04-07): fix broken RX outputs for remote config
fixes bug w/ broken "special pins" / incorrect channels outputs for RX when connecting remotely from TX to RX via Configurator.

partially resolves #206
PR #210 (2018-04-09): fix broken RX->TX serial bridge connection
- Add missing rx_reset() call to fix broken telemetry packet handling on TX side. 
- Without rx_reset() call, the received packet handler routine will incorrectly run on each loop iteration, which breaks the incoming serial packet handling.
- this resolves #206
closed 2018-04-09 by csurf