openLRSng/openLRSng #207: Need a little help identifying some components for a 6CH rx

Rusty105, 2018-04-03:
Hey guys, working on one of these 6ch receivers https://oshpark.com/shared_projects/Hk0Ae7U4  and have most of the BOM figured out, I think, 
But need 3 items:
D1 Schotkey Diode
D2,D3 LEDS (red, green)
Y1 16MHz Crystal

The LEDs are 0805 size, but not sure about the others. Any help would be appreciated, thanks!

csurf, 2018-04-04:
@Rusty105 
I'm pretty sure that's the DTFUHF 6ch.
Easier/better if you just open the original design files. You'll need to download&install diptrace (free), and then get the original schematic & board files from the brotronics repo on github.

https://github.com/DTFUHF/DTFUHF-6ch
https://diptrace.com/

Rusty105, 2018-04-04:
Yeah, did that, I have the values for most, but not the schotkey, And i think I will just have to measure the pads for the crystal.  LED I can wing it, I think, just don't want too dim or bright

csurf, 2018-04-04:
for the diode, I just used whatever I had on hand; I believe they were 0.5A 30V or something. It just needs to be reasonable and of the right size (0805). Same for the LED's; they were 0805's. Any general one should do fine. The current will be limited by the inline resistor, and the voltage shouldn't make a difference in this case. Don't worry, they'll be bright enough...

Rusty105, 2018-04-10:
I think I got it. Thanks!

csurf, 2018-04-10:
ok, closing this out then...

closed 2018-04-10 by csurf