openLRSng/openLRSng #212: On/off telemetry direct by taranis

jack113113, 2018-04-12:
hello developers

Can we On/off telemetry easy by Taranis ?  then bind again ? in the fact , it is little inconvinience when us Configutor to this work

mikeller, 2018-04-12:
@jack113113: Can you please elaborate a bit what user interaction and resulting behaviour you would like to see implemented?

jack113113, 2018-04-12:
mikeller 

it same somthing here  , see photo 

[image]

mikeller, 2018-04-12:
This is already implemented if you use the multiprotocol protocol on the TX module. You can set output power (high / low), with the value of low set by the 'option' parameter, and just disabling telemetry in the telemetry menu of your TX will disable telemetry.

jack113113, 2018-04-12:
@mikeller 

Off telemetry on Taranis not display , but it does not Off telemetry from modul ,right ? 
i want to Off telemetry from modul .

if use configuration , then put it on Taranis with setup diffirent , so which it will work with setup  ?

jack113113, 2018-04-19:
i tried  to change option 0 to  7  in power RF , but maybe it not change . 
Old version Firmware taranis have not OFF  telemetry . right ? i do not see in menu

jack113113, 2018-05-03:
@mikeller  
i only can On/off Telemetry in ''' BIND ''  in internal Modul  Frysky Taranis . 
External modul Multi OLRS  , status '' Bind''  NOT choosen " ON/OFF TELEMETRY same frsky 2.4ghz 
I use OPENTX 2.2.1 x9d . I do not see where to OFF telemetry with external modul 

Can you support this problem ?

jack113113, 2018-06-04:
@mikeller 
i real want to OFF telemetry direct On taranis by Bind  that do not need use Configulator . I want to better range .

mikeller, 2018-06-04:
@jack113113: I still fail to see how and why you think that disabling telemetry on the TX will give you better range?

jack113113, 2018-06-04:
😢😭😭😭😭

geofrancis, 2018-06-13:
@mikeller disabling telemetry doubles the rc update speed, this allows you to use a lower radio rate giving you longer range. back when this was the "undead mod" for maximum range it ran at 8ch @ 4800 with no telemetry giving up to 8km on 100mw hardware.

mikeller, 2018-06-14:
@geofrancis: Disabling telemetry **on the RX** does. Disabling telemetry on the TX will not affect how the link works at all, since the TX only reads the telemetry data.

geofrancis, 2018-06-14:
yes thats the point he is getting at, the ability to disable the telemetry from the receiver from the handset. via the binding menu.

mikeller, 2018-06-16:
I still don't understand. What you call the 'binding menu' can be used to change settings on the *TX*. This is different to settings on the *RX*. If user want to perform configuration of the RX, (or change all of the settings on the TX) from the TX, then then they should write a lua script that implements the configuration function of the openLRSng configurator.

geofrancis, 2018-06-16:
but telemetry is set by the Tx module, then the receiver gets those settings during binding.

mikeller, 2018-06-16:
@geofrancis: Please read https://github.com/openLRSng/openLRSng/issues/212#issuecomment-397774665. Changing the TX settings through the serial port is already possible, it just has to be implemented in OpenTX.