openLRSng/openLRSng #213: openLRSng Roadmap??

Rusty105, 2018-04-12:
First off, thanks for taking this back up. I really like this system and glad to see it get some legs again. With that, what are the plans for openLRSng? From what I have been reading, a big code clean up was/is underway. After that what are the future goals?

Rusty

csurf, 2018-04-12:
@Rusty105 
controversial subject. I had lot's of ideas, but getting support & approval has been a limitation. I'm personally ready to fork & part ways, so I'll definitely be keeping things active on my end (just an FYI, in case this might be of interest to you).

Rusty105, 2018-04-12:
This is going to be a dumb question, But who is "in charge"? lol.....

mikeller, 2018-04-12:
My immediate intention for this project is to take the steps necessary to ensure it's availability for the future. With the impending end of its main distribution channel (Chrome web store), and being all geared towards leveraging this distribution channel, this step alone requires considerable changes in the release process. And that is just to preserve the status quo (users can install an app and then use it to update the TX / RX firmware, and configure it).

After that I am open to suggestions. Taking into consideration the large user base, and the range of available hardware, I think it makes sense to have 'ongoing support of openLRSng on existing hardware, and retention of interoperability with new hardware' as a goal in this project. If developers want to go entirely new ways, then doing so in a fork is the right way in my opinion.

Re who is in charge: All of us really, this is a community project.

Kelly-Foster, 2018-04-12:
I generally agree with Rusty's suggested improvements, and I appreciate the energy that @mikeller and @csurf have recently invested in the rejuvenation of openlrsng.  I am bummed it became political/contentious fairly quickly, and will track @csurf's efforts if he forks off on his own.  FWIW, I don't have huge concerns about interoperability, and would happily invest in new Tx/Rx hardware and flash latest versions if it gave kick@ss range, reliability, bi-directional telemetry, etc.  I am also eager to see a 32-bit based hardware platform.

Kelly

Rusty105, 2018-04-12:
I guess the major issue right now is as @mikeller  says, the Configurator needs to be moved to a standalone install (my words), so that when Google drops it users can have a program to use. I personally have no clue how to do that, my programming is somewhat limited. 32 bit HW would be nice, but until the existing chip is maxed out, it might be a big bite to try and digest right now, although it might make some of my ideas easier.  I'll help where I can.....

atomiclama, 2018-04-12:
Just thinking aloud here. 
Does betaflight have support for any I2C attached displays ? Never looked into this side.

fiveangle, 2018-04-12:
Yes, 32-bit would make adding things like integrating SSD1306 OLED display, comprehensive telemetry return packet analysis and general link quality gauging with auto protocol range escalation, and so many other potentially new autonous features easier ( since they wouldn’t require massive investment in optimization to run on the relatively modest AVR MCUs)  that it really will be a necessity to do so at some point if one expects any level of success in moving the “system” into the future.

However, as Mike and many of us have pointed out, doing so at the detriment of the very large current install base just doesn’t make a whole lot of sense.

A common approach used in other OSPs  transitioning from AVR to more modern MCUs  is a steppingstone approach to tackling the problem:  identify and or build 32-bit hardware that implements the existing codebase through a Hardware Abstraction Layer ( and corresponding jump in major revision level, drawing a line in the sand where support for new hardware begins).  

But with this approach comes the added effort of putting a lot of forthought and time into developing an adequate HAL that can cover any new hardware someone might wish to add.  It also  often means a small level of stepping backwards (or at least temporary feature stagnation), while the new HAL is vetted out.

Because someone disinterested in  maintaining backwards compatibility with existing install base would obviously not see value in all of that extra effort, I agree  that in such a case forking to a new project is the only real viable option.  And I don’t think anyone here is opposed to that.

The only contentious part is the notion that just because there isn’t anyone immediately available with time and interest to spearhead the HAL approach, does not mean that we should accept cannibalizing the existing project just for the sake of getting it onto more modernized hardware.

Rusty105, 2018-04-12:
@atomiclama 

Coding skills?  lol.... I can make an arduino blink a LED.  ok, maybe a little more than that that, but definitely not at the level of this project. I can maybe run tests, online docs, user how-tos. Right now tinkering with some rx hardware. All that said, I can pick it up fairly quickly if I dedicate some time to it. Which is one reason for the roadmap, to figure where we are going....

@fiveangle I don't think we have to abandon current users to move forward, but I also believe that forking the project, just to advance hardware, is not the way to go (or is a last resort only) it only dilutes the already thin participation.  Dragon link is now at V3, I think, and they didn't change their name...  "openLRSng" is not, in my opinion, just a ATmega328 and a RFM22, it is sort of branded at this point.

Hardware changes are inevitable, but just like Ford stopped supporting the Model T, this project would need to stop supporting a 8 bit microcontrollers, at some point. When?  Not really for me to say, don't know the history or the code well enough, but from what I have read so far, and seeing what some others are accomplishing with the same/similar hardware, I don't think we are there yet. Maybe it could be done in stages? Upgrade the microcontroller first. It is my understanding that we are just sending the data to the RFM22 and the RFm22 is doing the packet packing, and unpacking. Now I may be naïve, but don't see why a 32 bit tx couldn't communicate with a 8 bit rx, and vice versa, if they are using the same RF module. You would just be feature limited by the old hardware. If the 'new' TX had a legacy mode, existing users could keep the older RXs installed

HAL? Does it make sense on an 8 bit microcontroller? Wouldn't it add overhead? Maybe when the project moves to 32 bit?

User base...  Just how big is the user base? More important, how many are active?  I did a quick YouTube search for "openLRS"  And not many videos. Ok that is not the most scientific method, but I was only on the second page, and quickly found myself watching videos >3 years old. products like Dragon link and Eagle Vector have pages of recent videos. Again, not scientific.

mikeller, 2018-04-12:
@atomiclama:

> Does betaflight have support for any I2C attached displays ? Never looked into this side.

Yes it does - but not sure if any of the current targets actually include support for it.

Some more thoughts re backwards compatibility. To me, the hardware that defines openLRSng is the RFM22 (and relatives). This is pretty much the defining element of the radio link, and will be no matter what MCU is driving it. I don't think that changing to a more capable MCU family while still using the RFM22 will allow us to wring significantly more out of the radio link. To me, this also means that, even if the project moves on to support another MCU family, it will be worth to preserve interoperability with the existing hardware at the radio link level, so that users can mix new and old hardware.

Kelly-Foster, 2018-04-12:
Re @atomiclama query about assistance, I could help QA and flight test a system that was headed in the direction of Rusty's and my requests.  Former C developer, now with a good Ubuntu setup under WSL, with the ST debug gear NIB and ready for action.  Plenty of fixed wing's in the hangar, currently using a mix of 5.8G and Dragonlink for various FPV systems.  Little spare time, but plenty of interest, and some relevant experience.  ;-)  Big fan of ArduPlane, Tridge and the boys, and open source flight control.  Several disposable foamies for high-risk alpha testing.  

Kelly

fiveangle, 2018-04-12:
> Now I may be naïve, but don't see why a 32 bit tx couldn't communicate with a 8 bit rx, and vice versa, if they are using the same RF module. You would just be feature limited by the old hardware. If the 'new' TX had a legacy mode, existing users could keep the older RXs installed

That's the contentious part. Most proposals are to abandon compatibility with the current implementation for sake of moving to more capable strategies and RF links (@csurf's "long range" mode is just one example).  Most are all on board if the extra effort to support backward compatibility with current user install base is maintaned and to not allow changes where backward campatibility would be fully comprimised.

> HAL? Does it make sense on an 8 bit microcontroller? Wouldn't it add overhead? Maybe when the project moves to 32 bit?

That's the whole point... the 8-bit MCUs incur zero additional overhead as their HAL becomes essentially a pass-through, and more capable MCU equipped modules incur only the minimum required runtime overhead.  Most of the HAL for such a simple project as this (relatively speaking) can easily be easily handled at the preprocessor level.  Of course this is just the one approach I've observed used in other projects with success but certainly isn't the only way.

-=dave

Rusty105, 2018-04-12:
OK, in regards to "long range", or other modes,  why does it have to be a "this" or '"that" Why not just give the user a choice at firmware uploading time. Let them decide what 'flavor' to upload.  For example, Ardupilot has several major, vastly different, 'modules' -copter, -plane, -rover, -sub, -antenna tracker, (may have missed one), yet they all use the same interface, Mission Planner, for firmware loading, configuration, and operation. The openLRSng configurator needs an overhaul to get off Chrome apps anyway, add some features that allow customization of the tx/rx firmware.....
Splitting into fragmented projects just fractures a user base, leaving each project fewer users and less interest. I don't think anyone really likes to have several systems to cover all of their needs. if users could use the same system, openLRSng, for all their RC control/telemetry needs, and just pick and chose what options to enable, I think it would make for happier, LARGER,  user base......

csurf, 2018-04-12:
@Rusty105 
ardupilot has always been a project that's very open to evolution and development of new features, even if the cost of doing so is to phase out & deprecate old features in favor of new ones. They've also always had the advantage of having more resources to work with (ATMega2560 from the very beginning, which they milked every last drop out of). We're stuck with '328 which has always been a big limiting factor.

Having different firmware images based on a desired feature-set was an idea I also had; to have different builds that you could download based on what you needed. Something like having a 'mavlink build', a 'gps-enabled beacon' build, a 'long-range mode' build, a standard/legacy build, etc. The problem is that the codebase would get very messy very fast unless things were reaorganized in a really effective way so that it would be easy to generate different builds without having it turn into a tangled mess...

I cringe every time I see & hear mention of 'backward compatibilty'. It's a completely false requirment and limitation as there's actually no such thing; you can't even bind across firmware versions (major & minor version must match in order for the modules to bind). Besides, why would you even want to do run different firmware revisions when you can simply load the desired firmware on both the TX and RX? It isn't that difficult to backup & load a particular configuration and firmware image.

I simply don't understand the resistance to moving the project forward with new, improved features. It's been left behind by the other *LRS systems due to its lack of progression. People are fighting to keep things stuck in a rut in the name of 'backward compatibilty' instead of trying to work on improving it. Plus, there's nothing holding a user back from simply running an older firmware image if they don't approve of using newer firmware with different features. 😕

csurf, 2018-04-12:
@Rusty105 
the bind versions (major & minor version values) must match. it's built into the code.

I'm overhauling my personal fork, and will publish when ready. It'll be based largely on the current code, but with lots of changes based on my personal coding style and way of doing things. I also have to rebuild/overhaul the configuration app, which basically means rebuilding a bunch of stuff, and then restyling it so that it looks different. I'll have to come up with some kind of brand/name/logo to distinguish it from openLRSng. Needless to say, there's still a fair amount of work to do until I'm ready to publish...

mikeller, 2018-04-12:
@csurf:

> Plus, there's nothing holding a user back from simply running an older firmware image if they don't approve of using newer firmware with different features. ￼

Are you volunteering to provide support for these users?

Rusty105, 2018-04-13:
@mikeller 
but if can keep some form of interoperability between tx/rx, wouldn't it be easier? Say instead of having the Major and minor version needing to be the same, how about only major version, and have it so that as long as the tx has a higher major, the system should work.

I am just spit-balling here, trying to find a compromise between moving forward and retaining some "backwards" compatibility. Maybe the above wont work,????

mikeller, 2018-04-13:
@Rusty105: I wasn't aware of this hard limitation of only allowing RX and TX with the same major / minor versions being allowed to interoperate. This is certainly something that we'd have to look into for versions going forward, and probably replace it with specifiers for a 'minimum required version' on both sides, and then have both sides check the remote side's reported version, and if they meet the requirements, allow binding.

Rusty105, 2018-04-13:
@mikeller , @csurf mentioned it above, not sure where to look, maybe in "binding.h"

atomiclama, 2018-04-13:
> Yes it does - but not sure if any of the current targets actually include support for it.

Ah cool that will make things a lot easier getting a display up and running

mikeller, 2018-04-13:
@csurf: I now know it's there. I am just not sure this is done in the way it should be, since it's pretty arbitrary.

mikerc2017, 2018-04-13:
anyone know  why  @kh4 stop develop Olrs ?

Rusty105, 2018-04-13:
So, even after all this discussion, what is the plan? ......

kh4, 2018-04-14:
I ceased mostly due to lack of time to work on this project. Also I currently do drones for work so there is a potential conflict of interest from that side too due to e.g. NDAs towards vendors. 

I am still very fond on this project, but I need something else to freshen my head after work, I trust many of you can relate to that.

Best regards, Kari (kha)

Rusty105, 2018-04-14:
@kh4 
Well.... Thank you for getting it this far.

So.... How would you like to see it progress?

Rusty

mikerc2017, 2018-04-15:
invite Midelic on rcgroup to intent this project ? i believe OLRS will develop again if he work with it

mikeller, 2018-04-15:
@mikerc2017: What exactly do you want to ask @midelic to contribute to this project? I think his area of specialisation is the FrSky protocols, and I don't know how this could help us here?

Rusty105, 2018-04-17:
So....

What should we look at first?  Configurator? get that off Chrome apps so users have a stable configuration tool.

mikeller, 2018-04-17:
Configurator needs a couple of things done before this can happen in a user friendly way:
- add support for checking GitHub for newer configurator releases, and display an update prompt to the user (done);
- add support for checking GitHub for available firmware versions, allow the user to select the desired one, then download and install it;
- convert the app to NW.js (or equivalent).

Rusty105, 2018-04-18:
I was poking around the net trying to find programs that help write standalone apps that could be used across multiple platforms and found "Qt"  https://www.qt.io/   Runs on Win , Linux, and OS X, and can compile apps for the same and then some. Could using this SDK, or something like it,  be an option?

Rusty105, 2018-04-18:
@mikeller OK, I will take a look.
I am not 100% familiar with what is available out there, but if giving a direction I usually can find my way.

mikeller, 2018-04-18:
@Rusty105: Have a look at https://github.com/betaflight/blackbox-log-viewer/pull/88 to see what had to be done to convert another Chrome web app to NW.js.

Squidyman, 2019-04-16:
Has this turned into much? I've been following OpenLRS a few years and have used the software off and on. I'm a C/C++/Java developer who'd be happy to help bring OpenLRS forwards. Is there an active forum anywhere?

jonathan2018, 2019-04-19:
> Điều này đã biến thành nhiều? Tôi đã theo dõi OpenLRS một vài năm và đã sử dụng và tắt phần mềm. Tôi là nhà phát triển C / C ++ / Java / Python / etc, người rất sẵn lòng giúp đưa OpenLRS về phía trước. Có một diễn đàn hoạt động ở bất cứ đâu?

Yes. i hope  someone keep work OPENLRS  .Do you try modifi code by use STM32 ?

Squidyman, 2019-05-13:
> Yes. i hope someone keep work OPENLRS .Do you try modifi code by use STM32 ?

I haven't looked into STM32. Venturing into embedded electronics is fairly new for me. Are there any active developers working on the project? It sounds like OpenLRS has reached it's limit with the current hardware.

fiveangle, 2019-05-23:
The GitHub tracker, for all is faults, is where communication is happening. 

The software is still working perfectly on my setup so while there hasn't been much action, they're isn't much needed. There had been discussion of protocol additions to gradually degrade latency as a function of packet loss, but other than that lacking (compared to Crossfire for example, which somewhat needs that feature in order to compete in range with the more penetration 433Mhz signal of other long range systems).

OpenLRS is still as stable as it always has been, but the project is more than open to any feature improvements that don't break backwards compatibility !