openLRSng/openLRSng #214: PPM output not compatible with iNAV

padcom, 2018-05-05:
I have tested a number of different receivers outputting PPM with iNAV and Omnibus F4 v2 board:

- https://www.banggood.com/Tiny-Frsky-8CH-Receiver-Compatible-With-FRSKY-X9D-Plus-DJT-DFT-DHT-For-QX90-DIY-Micro-Quadcopter-p-1079963.html
- https://www.banggood.com/DIY-FRSKY-8CH-Receiver-RX-PPM-Output-For-X9DPLUS-XJT-DJT-DFT-DHT-p-987247.html
- https://hobbyking.com/en_us/dsm2-6ch-without-case-and-no-cert.html

In all cases PPM output gets properly through to iNAV (tested with 12 channels).

OrangeRX LRS receiver with OpenLRSng is different and does not get through properly.

I know there is a jumper that can be removed to get the OrangeRX LRS receiver to accept PPM from OrangeRX LRS receiver but can't seem to find the reason why the other PPM receivers do work.

Is it possible to get the LRS receiver to work?

geofrancis, 2018-05-05:
you need to limit ppm channels to 8 in the rx configuration page.

padcom, 2018-05-05:
I've done that and I have disabled injection of RSSI on channel 12 (just to make sure there aren't any additional settings that might somehow affect it). iNAV still doesn't see the PPM signal from the RX.

It is worth mentioning that the 8 channel limit is not needed when the jumper is off - too bad it needs to be soldered :(

padcom, 2018-05-05:
I have tested it again on 2 other OrangeRX LRS receivers and the situation is all the same. I wonder if this isn't some kind of design flaw on the receiver. But if that would be the case then why removing the jumper suddenly makes the PPM signal recognized by iNAV?

mikeller, 2018-05-05:
I know this isn't an answer to your question, but why aren't you using SUMD if you want to control iNav over openLRSng? This will give you 16 channels without any issues.

padcom, 2018-05-05:
Quite frankly I am as this is the only viable option at this time. I just can't let go of the fact that all other receivers do work with PPM. The number of issues flying around on both OLRS and Betaflight/Cleanflight/iNAV is huge and they all are being answered with "it just doesn't work; remove the jumper".

The problem with removing the jumper is two-fold: a) it hinders the ability to use SBUS all together and b) it requires intervention with a soldering iron which voids the warranty of the board which is bad.

I think I would let it all go if all the other receivers wouldn't work too but they are working fine.

I have one more test to do with another OLRSng-powered receiver (Wolfbox) but that needs to arrive and it will take some time. That way I'll be able to determine if it is the hardware that is causing PPM not to work or if this is indeed a software issue.

mikeller, 2018-05-06:
Fair enough. I don't know how much sense it makes to put more time / effort into this. Considering that the RX and the flight controller are both digital systems, using a digital protocol like SUMD to transfer information seems to be much more appropriate than using an analogue protocol, and putting up with limited options and inaccuracies.

padcom, 2018-05-06:
There is one more reason to pursue the issue: To connect the RX using SUMD you need a special cable that connects to the non-inverted UART1 using 3 pin JST SH connector instead of a regular servo wire. That connector is neither easy to come by nor long enough to mount the receiver on the wing without additional soldering. Also on the OrangeRX LRS receiver there is no direct connector available and the cable needs to be split between power line and separate UART port which is on the other end of the receiver. In both cases it makes for a far less durable connection than using a simple servo wire with PPM.

Unless there is a 3rd option (PWM doesn't count) I think it is really worth the effort to get to the bottom of it.

padcom, 2018-05-06:
I think I should have started the thread with the information that I am using the Omnibus F4 v2 on a fixed wing airplane rather than a multirotor where connections are not long and the individual components can be mounted next to each other.

geofrancis, 2018-05-06:
your setup should work, have you tested the ppm out with something else?

padcom, 2018-05-07:
Only with other receivers and the same Omnibus F4 v2 board with iNAV (see the first post). And that works. The only combo that doesn't work at the moment is OpenLRS with iNAV.

padcom, 2018-05-07:
I have good news (well, sort of...)

I have successfully tested iNAV with an OpenLRS receiver (Wolfbox). All the settings are the same as with the OrangeRX LRS receiver but this one just works.

Does anyone has any idea what is so different between the 2 receivers?

padcom, 2018-05-07:
Or at least what should I check to figure out the difference? Any suggestions are very welcome!

padcom, 2018-05-07:
Well, I have full success. Even though it requires intervention with the soldering iron I am happy to report that the problem is with the receiver itself (OrangeRX LRS RX) and NOT with iNAV or the OMNIBUS F4 v2.

Long story short on the ORX LRS RX between pins and the MCU there are 1k resistors. On Wolfbox RX there are no resistors. Once the one on channel 6 (PWM5) is bridged effectively removing that resistor the PPM signal goes through properly to the flight controller!

I presume one could replace that 1k resister with some smaller value to limit the current but for now this will do fine.

Maybe it'd make sense to put this information somewhere on the wiki?

For now I am closing this issue.

closed 2018-05-07 by padcom

timsical, 2018-06-09:
Padcom! Please please could you explain how you solved this? I’m having the exact same issue, and it’s been driving a little nuts! I’m no engineer, so if you could give it in layman’s terms that would be awesome! 

Thanks for any help!