openLRSng/openLRSng #216: Telemetry and RSSI Value not properly transmitted

ameeuw, 2018-05-14:
Hi, I have the problem that while my receiver is injecting an almost full RSSI value (~1700) via an RC channel to the flight controller, while my transmitter is only receiving a really low value (~4) if at all.

In between the telemetry breaks down completely, but comes back at some point and shows a really low rssi value.

I tried it on a self built receiver as well as an OrangeRx model.

Any quick ideas what it might be?

ameeuw, 2018-05-15:
Nono, I mean just the tx-rssi value displayed in my Taranis.
I tried the current master branch and it works.
Thank you for your quick answer @mikeller

closed 2018-05-15 by ameeuw

mikerc2017, 2018-06-02:
@mikeller 
MAX RSSI  with telemetry FRSKY D on taranis is 124 . Howerver , in INAV it only 97% max . 
i have to scale RSSI to 110 . Set rssi_scale =110 . So it is normal ?