openLRSng/openLRSng #217: IBUS  on rx OPENLRS

jack113113, 2018-05-30:
i wonder , atmega on OLRS  can work  with a IBUS  output instead SBUS ?  
IBUS  is faster SBUS ? right ?

This is the testing result captured from one of the testing video:

Flysky i6X – 13.7ms
Turnigy Evolution – 14.6ms
Crossfire (on X10) – 19.5ms
Frsky Horus X10 – 31.5ms
Frsky QX7 – 36.3ms
Spektrum DX6i – 41.5ms

mikeller, 2018-05-30:
@jack113113: Yes, iBus output could probably be added. But the maximum possible update rate for RC commands is limited by the maximum data rate of the radio link, and it won't be any faster when iBus is used between RX and flight controller than when SBus or SUMD is used, so there is not much point in doing so.

mikeller, 2018-05-30:
@jack113113: That is correct. You can use a higher data rate on the radio link, at the cost of more errors / less range, but changing the RX - flight controller protocol will only give you a marginal reduction in latency, if any.

jack113113, 2018-06-01:
Yesterday , i tested R9M  of FRSKY and OLRS .  I can feel both same speed , latency same . 

however .with 500mw telemetry . R9M + rx Slim  ranger OLRS  and telemetry stable .