openLRSng/openLRSng #218: Failsave Sending wrong values when there is a recovery on a weak signal

jelle737, 2018-06-08:
Hi there,

I have three issues that I think are openLRSng related that caused a disarm of an aircraft because I suspect openLRSng sending the _wrong_ channel values.
The following video's should explain a bit what is happening:
https://www.youtube.com/watch?v=YUrNYp6kGyM
https://www.youtube.com/watch?v=fmgIClqqwgY?t=56s
https://www.youtube.com/watch?v=nbJEz2IkpHw

There are a few different possibilities that I suspect that could be happening:
A bad package is treated as a good package. But I suspect in the air packages are crc checked thus it it's ether good or bad and not in-between. 
Once a good package is received the ppm stream is updated mid stream, or cut of mid stream. That way the FC might get a mixed package and and up in a disarm state without a failsave state.
Once a good package is received there is however one or a few channels that receive a standard value before it's filled with a correct value. I suspect this because in the last two video's the camera pan's to the right, the cam is driven by a pwm signal.

In the third video we really went to reproduce the conditions of the first video (we picked a nice field to crash land), and it seems like we were able to reproduce it also.

With a hard change is signal (turn of TX, turn on TX) this behaviour is not (seems not to be) reproducible.

At this moment it's not the desired behaviour, it needs some fixing, but I suspect there are people here more familiar with the code.

Regards, Jelle

jelle737, 2018-06-08:
Going through the code, once going to failsave the failsave values are written towards the PPM channel array, this is done each channel at a time by halting and restarting interrupts. 
It seems like one received package contains all the data needed to update all channels. Here interrupts are halted before updating all channels and restarted after they are updated.

I think this means the ppm output stream could be / is updated mid stream. resulting in possible scenario a channel indicating disarmed status to a flight controller and a channel indication no failsave to a flight controller. (if it's updated in between those channels) However in video 1 the arm-disarm-failsave is all on the same channel.

Also the pan servo should not have moved as seen in video 3 (and 2), indicating a stray value. I see the openLRSng code does not have CRC or any value control. It could be that package correctness is warranted by the RFM chip but I believe a stray package can get through and might actually be what is happening here. Could this possibly be what's happening here?

Regards, Jelle

mikerc2017, 2018-06-09:
in your videoS , RSSI  is so low . 
do you have use telemetry  Frsky D on OPENLRS ?  or you only use RSSI  analog , inject channel ?

jelle737, 2018-06-10:
Ofc RSSI is low when flying on the edge or range. That is what failsave is for.
I do not use telemetry, the telemetry range is less and effects the overall use of the system, therefore disabling telemetry provides a more solid system. RSSI is injected into a channel, it's just RSSI, not LQ or any combination.

jnb81, 2019-01-13:
I tested 3.9.0 rc2 on my skysurfer (approx. 35 flights) and have no issues. I can fly 10km easily (with 60-70% of RSSI :) just love open lrs multi!)
https://youtu.be/Dz7UQQBEoPM
I use RSSI from ch1 and low pass filter to convert it to ADC value (after some adjustment/ calibration it goes well from 100-0). 
The only problem I had was when I lost signal during landing due to having transmitter very close to the ground and plane landed in fog 2-3meters under my sitting position (~300m from me). RTH (!FS!) squeezed the battery to death despite being so close couldn't hear or see (due to the fog) the plane. 
https://youtu.be/KHGQY3rfilI

jelle737, 2019-02-06:
Digging through the code, the PPM stream is created from data within the PPM[] table, while independently the received packages are decrypted and that PPM[] table becomes updated. Making one of my first thoughts true:
> Once a good package is received the ppm stream is updated mid stream. That way the FC might get a mixed package and and up in a disarm state without a failsave state.

If a failsave is set to output a certain failsave frame then on receiving new channel data the created frame can be a mix of the failsave frame and the new frame. Making disarm without a failsave possible (if disarm is on channel 4 and failsave is on channel 8 for example).
I'm basing my conclusion on the code I read here:
https://github.com/openLRSng/openLRSng/blob/2c5d3fc7970fe91caef852a409be46666c58beb8/openLRSng/RX.h#L263-L267

This might show that these days it would be better to not output anything on failsave (and have a flight controller handle it) or to have disarm/failsave triggers on just 1 channel (what I currently use).

It's really a once off very small chance but seems like it happened enough to have multiple videos where I fly, and even have the iNav developers implement a smoothing on ppm input streams.

For me I'm happy with finding this solution and knowing how to work around it. It might be an idea to make a copy of the PPM table and only update that table in between frames. Something like this:

```
ISR(TIMER1_OVF_vect)
{
  if (ppmCountter == 0){
    thisPPM = PPM; //this is pseudocode
  }
  if (ppmCountter < ppmChannels) {
    ICR1 = nextICR1;
    nextICR1 = servoBits2Us(thisPPM[ppmCountter]) * PWM_MULTIPLIER;
```

mikeller, 2019-02-07:
@jelle737: Well spotted. The way the code tries to avoid the problem you describe is by disabling interrupts while `PPM[]` is written to, like here:

https://github.com/openLRSng/openLRSng/blob/2c5d3fc7970fe91caef852a409be46666c58beb8/openLRSng/RX.h#L453-L456

unfortunately, this isn't done properly in the case of failsafe, where only individual channel writes are protected and this atomic, but not the whole block:

https://github.com/openLRSng/openLRSng/blob/2c5d3fc7970fe91caef852a409be46666c58beb8/openLRSng/RX.h#L630-L632

Try this, this should fix it:

[rx.zip](https://github.com/openLRSng/openLRSng/files/2840162/rx.zip)

jelle737, 2019-02-11:
Hmm, seems like I mentioned in my second post that I already used the arm-disarm-failsave on 1 switch. Could it actually be that one of the channels (PPM and PWM) could be elongated because an interrupt was delayed? Like video 3 showing the pan servo moving on the moment of disarm, maybe that pulse was elongated because of a delayed interrupt. Similarly a channel in the PPM stream could be elongated explaining why it still disarmed when failsave-disarm is on the same switch in video 1. But how do we tackle this problem them?
Maybe instead of disabling interrupts there is a bool updating set true while updating the PPM[] table? and use a copy of that PPM[] table, updated if that bool updating is false in the ISR that builds the PPM signal.
I was searching where/how the PWM channes are build/made/generated. Are these also with an interrupt? I couldn't immediately find it.