openLRSng/openLRSng #224: Running RX at 20mhz

artemen, 2018-10-17:
Recent versions of atmega 328p run quite happily at 20mhz, I wonder if there are changes to the code which need to be done prior to compiling RX code for 20mhz target.

ElectricRCAircraftGuy, 2018-10-17:
I can't answer your question unfortunately, but it surprises me you said "recent." I believe *all* versions of the ATmega328 can run at 20MHz. (The ATmega2560 for instance only runs up to 16MHz I believe, but the ATmega328 has always been able to do 16MHz.)

artemen, 2018-10-17:
16 yes, all 9f them can do but i believe the recent ones can do 20mhz stable. At least there are many 328pau ICs at most of the component sellers like digikey and mouser which have up to 16mhz max clock speed and many which have 20mhz max clock speed. Also, i've notice 20mhz external oscillator targets in Arduino IDE boards.

artemen, 2018-10-17:
I think that quite a few packets get lost/discarded by missing hop interval window due to miniscule deviations in timing, this could be happening in part due to oscillator frequency drifting with temperature changes especially on high output units 1W+.