openLRSng/openLRSng #230: Forcing Failsafe causes Rx to drop connection

Buzzliteyear, 2019-02-04:
Turnigy 9x flashed with 821r ER9x set to openlrs multi
futaba 433 orange100mw tx flashed with tx-2.hex and 1k resistor between PPM and Rx port
openLRSng 3.9.0-rc3
forcing a failsafe on the ground Receiver drops the connection and will not recover
tested with two different orangerx receiver modules
does recover with old openlrs 3.8.8 firmware

has done this 3 times in the air now so stopped flight testing for now
caugth once on video
red light stays on the Rx and will not recover until a power off/on

https://youtu.be/4RmMhdGzc5k?t=3m7s

Brett

mikeller, 2019-02-04:
@Buzzliteyear: And by 'forcing a failsafe' you mean pushing the button on the TX module?

Buzzliteyear, 2019-02-04:
no i reduce the RF power and walk behind my steel container,
and hold Tx so I lose packets/beeps for at least 3 seconds
same as if im flying out of range

Rusty105, 2019-02-04:
This may be a hardware issue. It seems there is another LRS system popular on RC Groups having the same issue with the recent batch of Orange RX  recievers not regaining bind after a failsafe.

Makes me wonder how much  open source code is in that closed firmware????

mikeller, 2019-02-04:
@Rusty105: Any pointers to where these errors are reported?

Buzzliteyear, 2019-02-04:
@Rusty105. remember its not dropping connection with 3.8.8 when I tried today.... same hardware
@mikeller . thanks heaps for your effort on this project

Rusty105, 2019-02-04:
@mikeller , Try here  https://www.rcgroups.com/forums/showthread.php?2037442-Ultimate-LRS-Hassle-Free-Full-Mavlink-16-Channels-Fly-More/page968  post 14509. It's not the first time it was mentioned, but it has been for a while. I am not sure they are 100% related, but very odd

jnb81, 2019-02-04:
I just tested my old Orange RX module (2012-5-14 pcb version) with 3.9.0 rc3 and I can confirm this Failsafe issue is real. 
RX tries hard to keep the link but after longer blockade (in my case more than 5 seconds) it gave up, and stayed in failsave even when close to TX (TX however looked as if the link is OK)
I had 1000uF 6.3V low esr capacitor connected on 5V input (just to make sure it's not due to power instability).

jnb81, 2019-02-04:
BUT If you set up beacon frequency it can bring RX back to business! (e.g. escape from a loop) :-)

Buzzliteyear, 2019-02-04:
@jnb81 : good work around,
just tested myself and it does recover but for some reason even tho I set Tx to low power and 
it was full solid beep I had to walk away 150m before beacon would activate,(total tx to rx signal loss)
needs testing on FC as when I had failsafe delay set I was getting Dis-arms

FYI: signal does recover on HawkEye 9 channel and ChippLrs receivers

Buzzliteyear, 2019-02-04:
@jnb81 : good work around,
just tested myself and it does recover but for some reason even tho I set Tx to low power and 
it was full solid beep I had to walk away 150m before beacon would activate,(total tx to rx signal loss)
needs testing on FC as when I had failsafe delay set I was getting Dis-arms

FYI: signal does recover on HawkEye 9 channel and ChipLrs receivers

jnb81, 2019-02-04:
I also tested it on my diy rx which is based on flytron/orange pinout and the result is the same, when it gets into a look the only way back is the beacon (after the 1st call: Beacon(deadtime+interval)).
https://photos.app.goo.gl/UZKgZhAT1huLug2AA

Rusty105, 2019-02-04:
So, an open source code, openLRSng, and a closed source code, ULRS, show the same bug on the same hardware???

jnb81, 2019-02-04:
I think HawkEye 9 is Si4432 based module while Flytron/Orange have RF22S modules...

Rusty105, 2019-02-04:
So what has changed between 3.8.8 and 3.9.0rc3??

jnb81, 2019-02-04:
Well pretty significant change from ppm 3.8.8 to serial multi-protocol 3.9.0 :)

Buzzliteyear, 2019-02-04:
I just did I field test with Flight Control hooked up to show FS/rx signal recover.
seemed to recover well
Failsafe delay :200ms
Stop PPM:500ms
Beacon Frequency :462.562 Mhz
One point .......when I turned transmitter off lost packet beeper did not recover even tho I had control
and recover of beep indicator did happen 50m up hill maybe when rx to tx link was re established

https://youtu.be/rrxKKRF1nPs

Buzzliteyear, 2019-02-04:
@Rusty105, I dont know shit here either but is this what your looking for
https://github.com/openLRSng/openLRSng/releases/tag/3.9.0-rc3

Buzzliteyear, 2019-02-04:
@Rusty105, I dont know shit here either but is this what your looking for ( I am "pmanu" on RCGroups)
https://github.com/openLRSng/openLRSng/releases/tag/3.9.0-rc3

Buzzliteyear, 2019-02-06:
@ mikeller, great stuff Mike, its the orangerx 100mw so its the RX-3.hex
Mike I see you have something to do with trademe, do you live in NZ, im in Kaukapakapa
Brett

mikeller, 2019-02-06:
@Buzzliteyear: Yeah, I am in Christchurch. :grinning: 

Here you go, realised I had to zip them anyway, so I zipped all of them:

[rx.zip](https://github.com/openLRSng/openLRSng/files/2835713/rx.zip)

jnb81, 2019-02-06:
Thnx Mike, I did a test on my diy RX (based on flytron/orange routing). 
It can recover from failsafe if the signal was lost for up to about 3sec, for 5 and more no recovery :(

jnb81, 2019-02-06:
I did the same test with 3.9.0 rc2 and RX recovered well 2times.

Buzzliteyear, 2019-02-06:
@Mike thanks heaps. I can confirm jnb81 tests with mine are the same on Orangerx 100mw
still no recover on failsafe with rx.zip file but does recover with 3.9.0-rc2 file
tested 10 times, but would like to do a longer range as when beep/packets are heard seems tx can still control rx . we are only hearing rx side of lose packets
Thats going to be hard to take you fishing and drink beer being so far apart.... ;-)
@jnb81......cheers
Brett

mikeller, 2019-02-07:
@jnb81: Thanks for testing.

> I did the same test with 3.9.0 rc2 and RX recovered well 2times.

So that means with the firmware I provided on the RX, and 3.9.0-rc2 it works as it should?

@Buzzliteyear: Thanks for confirming. Unfortunately I am not often up north, would love to take you up on that offer. :-)

jnb81, 2019-02-07:
@mike
By RX I meant RX module (in my case diy, based on RX-3 type - flytron/orange), sorry for confusion. 
So RX module recovers well on 3.9.0-rc2 (after failsafe), while on 3.9.0-rc3 and [rx.zip](https://github.com/openLRSng/openLRSng/files/2835713/rx.zip) RX module has problem to recover and stays in failsafe (even in area of good signal).
I had practicly the same results as [Buzzliteyear](https://github.com/openLRSng/openLRSng/issues/230#issuecomment-461169930)
>I can confirm jnb81 tests with mine are the same on Orangerx 100mw
> still no recover on failsafe with rx.zip file but does recover with 3.9.0-rc2 file
> tested 10 times,

Buzzliteyear, 2019-02-07:
@Mike............. Id Like to Update my FYI comment above

it seems time is a factor

Hawkeye Rx does not recover if in failsafe for 10+ seconds (AtMega328p chipset)
ChipLRS receiver Does recover after 30+ seconds same chipset onboard usb
both with 3.9.0-rc3 firmware
@Mike...Boat and beer waiting.....

jnb81, 2019-02-09:
> Hawkeye Rx does not recover if in failsafe for 10+ seconds (AtMega328p chipset)
> ChipLRS receiver Does recover after 30+ seconds same chipset onboard usb
> both with 3.9.0-rc3 firmware

Super weird ChipLRS is practically based on Flytron/Orange schematic.  
@Buzzliteyear: Does your ChipLRS receiver have LFCN filter?
https://github.com/axulige/ChipLRS/wiki/English-Receiver

Buzzliteyear, 2019-02-09:
@jnb81. yes I cut open wrapping to check and it does Rx V2.1 LFCN-490
But so does my Hawkeye 9Channel

Buzzliteyear, 2019-02-09:
[image]
[image]

jnb81, 2019-02-09:
@Buzzliteyear Great pictures!
Btw your ChipLRS receiver only has 0Ohm resistor instead of LFCN.  
Only HawkEye receiver has LFCN.
Very strange... that ChipLRS receiver is not affected...

Buzzliteyear, 2019-02-09:
@jnb81,thanks.....yes I did notice that and that was going to be my next part to research if there was such a LFCN-490 with those zero markings, But looking at it closer the Black chip has no ground pins on the side so has to be a resistor.
now I have to search my email to see what I did ask for/ordered
I also have a Flytron Rx but is inside a wing . its a nice red colour ;-)
Yes strange.

Buzzliteyear, 2019-02-10:
@Mike....testing today with 3.9.0-rc3 firmware and beacon On so failsafe does recover Has another Issue

in iNav there is a setting I use so Flight Controller stays in failsafe until sticks are moved to help prevent system flicking in and out of FS before link is good again
set failsafe_recovery_delay = 0
this feature does not work now and when link is recovered Flight Controller flicks out of Failsafe
maybe dew to the beacon looking for Tx and forcing a link again

jelle737, 2019-02-11:
@mikeller you provide a compiled solution, but there is not really discussed what could have been the problem in the code. Also it's unlikely that openLRSng and ultimateLRS have a related problem so it made no logic to link to uLRS rcgroups thread by Rusty105.

mikeller, 2019-02-11:
@jelle737:

> you provide a compiled solution, but there is not really discussed what could have been the problem in the code. 

I still have not worked out what the whole problem is - if you have a better solution then I'll stand back and let you propose a fix.

> Also it's unlikely that openLRSng and ultimateLRS have a related problem so it made no logic to link to uLRS rcgroups thread by Rusty105.

Do you care to explain what your stake in this is, or is this just your way to let people know that you make unsolicited statements and should be ignored?

jelle737, 2019-02-11:
In both cases I'm curious and find it peculiar that both openLRS en uLRS show a similar problem. Since I dug through the RX code a bit an unresponsive RX makes no sense. And if something would be broken somewhere making the code halt then the watchdog should kick in making the RX start over? But you seem to at least have an idea. While I have none.

mikeller, 2019-02-11:
@jelle737:

> In both cases I'm curious and find it peculiar that both openLRS en uLRS show a similar problem.

Unfortunately, bugs propagating from open source projects into self proclaimed closed source 'projects' that are in reality just ripping off open source developers is all to common, so 'peculiar' isn't how I would describe it if exactly the same regression error is introduced in 2 projects around the same time.

> But you seem to at least have an idea. While I have none.

I am just looking through the refactoring that has been done, and looking for regressions that were introduced. I have found one so far, but I first need to set up a test environment to be able to confirm that this is what is causing the problem.

artemen, 2019-08-14:
I have experienced something very similar recently 3 times in a with dtfuhf 1w delux (modded to around 1.3watt) and diy broversity running 3.8.8,  after failsafe in challenging rf environment (right above railroad comms tower around 7km away 30meters above ground)  RX remained in failsafe for about 2min even while well within previously good signal range.  not sure if that's related to the discussed issue or was RX's rf front end overloaded by a high output UHF a few meters away (5-10meters), regardless the same behavior repeated 2 more times during the same flight, but much further away from the original spot.

mikeller, 2020-04-10:
@Buzzliteyear, @Rusty105, @artemen, @universam: I gave this another shot, but could not identify what is causing the endless loop without beacon.

In order to get movement on this, and be able to release a new version with all the other improvements, I have reverted the RX code refactor. Can you guys please test this RX firmware and report back if it works, and fixes the failsafe lockup problem:

[openLRSng_RX.zip](https://github.com/openLRSng/openLRSng/files/4460169/openLRSng_RX.zip)

jnb81, 2020-04-10:
Hi Mike, 

I did a test with [openLRSng_RX.zip](https://github.com/openLRSng/openLRSng/files/4460169/openLRSng_RX.zip) on my Orange UHF RX module and result is the same, RX won't recover after lost of signal. 
I appreciate how much energy you put here. I still fly with openLRSng
I use 3.9.0-rc2 on RX and 3.9.0-rc3 on TX tens of flights many times being out of signal, FS, works very well. I love it (RX recovers very well in this setup). 
I think only @csurf is able to fix this, as he made the last changes, I'm not sure what's with him ;( 

> @Buzzliteyear, @Rusty105, @artemen, @universam: I gave this another shot, but could not identify what is causing the endless loop without beacon.
>
> In order to get movement on this, and be able to [...]

mikeller, 2020-04-10:
@jnb81: Thanks for the quick turnaround testing. Can you please test with this:

[openLRSng_RX.zip](https://github.com/openLRSng/openLRSng/files/4461265/openLRSng_RX.zip)

jnb81, 2020-04-10:
> 
> 
> [openLRSng_RX.zip](https://github.com/openLRSng/openLRSng/files/4461265/openLRSng_RX.zip)

I tested this fw, I'm having bind issues. RX is accessible through Configurator, but cannot bind with radio (TX).

Buzzliteyear, 2020-04-11:
Sorry for delay I had to re-configure my gear,
I can confirm with jnb81 .....But mine shows in the Rx leds that it has bound
 but when re starting Tx and Rx they are now Bound

mikeller, 2020-04-11:
Okok, I'll have through my box of old gear and pull out the openLRS kit - I need to get set up to locally debug to get to the bottom of this.
Last time around the problem was that I could not think of a good way to simulate the original bug (RX not coming out of failsafe) on the bench. Any suggestions for how to do this?

Buzzliteyear, 2020-04-11:
ps. edit spelling misstake sorry...."NOT BOUND"

Buzzliteyear, 2020-04-11:
@mikeller  are you asking how to setup on the bench ?
if so set Tx to low power "0"  with telemetery on so you can here the packets drop
then walk away with Tx atleast 100m behind a wall and position tx in such a way that you force a continue beep for at least 10 seconds then walk out to the open to  try recover signal.

jnb81, 2020-04-11:
My test setup:
https://youtu.be/l6OqM6FuNds
I'm using Faraday cage aka a cooking pod with aluminum foil on the top :) 
With this setup you can have tx (set power to 1) and rx 2m away :)

jnb81, 2020-04-11:
UPDATE,
I was wrong about the pod! I'm sorry, 
You really need to go further (rx recovers well (2m from tx in a pod) even being longer in FS), I also have a concrete garage with a big metal door so that's how I usually test this.

I changed tx module to my diy and rx recovers nicely with this fw
[openLRSng_RX.zip](https://github.com/openLRSng/openLRSng/files/4460169/openLRSng_RX.zip)

But not with orange ones. 
@Buzzliteyear 
Did you test this FW?
What tx module do you use?

jnb81, 2020-04-11:
Very strange 
I have about half a chance of recovering and not recovering with orange tx module and fw
https://github.com/openLRSng/openLRSng/files/4460169/openLRSng_RX.zip
4 tests, 4 binds and 2x success 2x not recovering.

Buzzliteyear, 2020-04-11:
@jnb81  yes if that was the last file mike uploaded in this post.as they all have the same name
I use Orange Tx and Orange Rx
If you do your test in steel bowl again but make sure the bright light you see in the video stays on for 10+ second that might work....the longest i counted was only 5 seconds
,

jnb81, 2020-04-11:
Hi, 
I was unable to repeat the issue in a Faraday cage.  From 30 tries only 2 positive (stuck in FS). With fw RC03 on both.
I changed tx, battery, binds, ...nothing. 
Then (after 2 hours) I turned to my old method, blocking the antenna with RF shielding cloth fabric and I always can repeat the issue. (Orange TX (RC03) with Orange RX RC03, no telemetry, full power 0.1W, blocked through 2 brick walls + shielding cloth)
https://youtu.be/lT_lw0dlolM

jnb81, 2020-04-12:
Hi All, 

Couple of notes from my further testing:

- aluminum foil can be used as a replacement for RF shielding fabric cloth (works similar, it only quickly deteriorates, be careful to not short your RX!, [isolate the pins](https://drive.google.com/file/d/129-B61MDdRCeHWfs_v2_lmsdZKgocH-3/view?usp=sharing)) You need to play with covering the antenna and module together with finding the "right" spot. 
- the issue seems to be rather related to RF noise or signal distortion than being off the signal (as the RX within the Faraday cage almost always recovers), Now I understand some people couldn't spot the issue.   
- TX power seems to have little effect if any, I did try 0, 1, and 7 and I haven't noticed difference. 
- My [DIY TX](
https://drive.google.com/file/d/1vvHdQiAoPBROb00T-ap2cRazMoK04paS/view?usp=sharing) flashed (based on Flytron/Orange design) with 3.9.0 RC03 recovers well with RX on 3.9.0 RC03. I couldn't repeat the issue with this setup. I did try cloth for more than 30s, foil, Faraday cage, still recovers.  It's a bit of mistery to me. 
The only difference (with my orange tx) is that feedline coax is soldered directly to RFM22 module, and then that I used 3 LDOs (5V (as auxiliary (dropping from 3S battery) from a radio), 4V( for atmega) and 3.3V for RFM module (with 1kOhm resistors on SPI))
So the only difference no SMA joints, and different voltage levels.

Buzzliteyear, 2020-04-12:
I think the faraday cage is to close to Tx, you need distance. 
the signal only need to recover for a micro second and you can miss that point even with second led on
where as at a distance it is truly dropped complete connection

jnb81, 2020-04-13:
Hi Brett, 

I did another test, in which I wanted to demonstrate 3 cases (fw rc03):

- 1st RX in Faraday cage, antenna polarization matched with TX ( RX goes to FS and recovers easily)
- 2nd antenna  blocked with battery, polarization changed, failsaving even without cover, with cover constant FS, despite that RX recovers. 
- 3rd find a spot without cover when RX stays in FS, surprise RX won't recover...

https://youtu.be/cpkyaT8P9Qg 
This test can be repeated, make your own conclusion.

jnb81, 2020-04-13:
I cannot repeat this with my 100mW [DIY TX](https://drive.google.com/file/d/1vvHdQiAoPBROb00T-ap2cRazMoK04paS/view?usp=sharing) (practically the same design as Flytron/Orange)
RX recovers well.

jnb81, 2020-04-15:
I repeated the test with rc02 on RX the same setup (both orange modules)
RX recovers whatever I try. 
Very strange.

Buzzliteyear, 2020-04-15:
yes still a problem with 3.9.0-rc3.