openLRSng/openLRSng #231: New installation of configurator, how?

Rusty105, 2019-02-18:
I need to install the configurator on a new Win 10 Laptop, what is the recommended route?

Rusty

mikeller, 2019-02-20:
@Rusty105: Currently, the best way is still through the Chrome web store, as described here: https://github.com/openLRSng/openLRSng-configurator#installation

cshunt, 2019-02-23:
I was able to install the 3.8 version from the web store but it doesn't look like the apps have the same shortcut that chrome once did, you need to go to chrome://apps/ after installing to see them to launch...
using nwjs is the non chrome method.
http://goo.gl/iX7dJx
chrome://apps/

mikeller, 2019-02-23:
@cshunt: You should be able to create a bookmark / shortcut from the one you find under chrome://apps/.