openLRSng/openLRSng #233: LGT8F328P @ 32MHz

artemen, 2019-11-21:
hi, according to many people LGT8F328P  behaves exactly as atmel328p with a benefit of being able to run at 32MHz at 3.3v. 

twice the clock and twice the EEPROM size plus an extra timer (0/1/2/3)
i would like to give it a try, however, there is some timer voodoo going on and need help setting proper prescalers, 
if someone can take a look at this [datatasheet](https://github.com/dbuezas/lgt8fx/blob/master/docs/LGT8FX8P_databook_v1.0.4.en.pdf)
 It is easy to do CTRL+F for timers (same naming as atmel328p), however the issue is, I am not sure which ones are used throughout the project and here i need help of someone more intimately familiar with openlrsng codebase. 

AFAIK faster and more precise main loop is always beneficial, maybe it can even lead to improved packet loss rate.  there ARE packets dropped even with perfect signal strength, which could be a sign of some packets being discarded due to slow main loop. 
 
and maybe some type of checksum based packet recovery could be possible with the added clock-speed eeprom prog-memory size? 

there are hundreds of 328p based boards out there, it would be so simple to just replace one MCU (less than 50cents)  and do a firmware upgrade and get improved range and maybe a few extra features. 

https://github.com/RalphBacon/LGT8F328P-Arduino-Clone-Chip-ATMega328P
https://github.com/dbuezas/lgt8fx

thanks!

mikeller, 2019-11-21:
@artemen: Yeah, there is definitely potential in porting openLRSng to a more powerful MCU - especially if backwards compatibility is retained so that users can keep using the RX that are already in their craft.
Not sure how much help you can get in here - there is not much development activity in here.
Maybe open a pull request for adding support for the LGT8F328P (even if it is not completely working yet), so that you can ask more concrete questions about things that don't work?

artemen, 2019-11-21:
at least I was able to compile the code for RX by adding to RX.h

this: 
#if (F_CPU == 32000000)
#define PWM_MULTIPLIER 4
#define PPM_PULSELEN   1200
#define PWM_DEJITTER   64
#define PPM_FRAMELEN   80000

and this: 
#if (F_CPU == 32000000)
      TCCR2B |= (1 << CS22) | (1 << CS21);      // prescaler = 256

my lgt8f328p is arriving early December, so can't test at this moment, yet I am still  in the process of searching through the code whenever prescalers are used.