openLRSng/openLRSng #234: Question: SPI interface definition

hjchjcjh, 2019-11-22:
Hello everyone
        I am making a personal version of the hardware for Openlrs, but the SPI interface definition is different from any of the "hardware.h"
        So I am having trouble modifying the interface definition, because I don't know how to modify the hexadecimal content of the markup in the figure. Can you tell me where I can understand the meaning of these hexadecimals? thank you very much

[image]

danielsmd, 2019-11-25:
Hello.
I'm working on improving the Gitsly branch to work with STM32F103. 
I first started with modifying the SPI to use at higher speed, but without an overall improvement in transfer speed (I'm focused on telemetry with Mavlink).
Here are my modified project files:
https://github.com/danielsmd/OpenLRS-Gitsly-HW-SPI

mikeller, 2019-11-26:
@hjchjcjh: These are bitmasks for the respective IO ports:

`(x & 0x04) == 0x04`: check if bit 3 (`0x04` = 00000100) is set
`|= (1 << 4)`: set bit 5 (`1 << 4` = 00010000)
`&= 0xEF`: reset bit 5 (`0xEF` = 11101111)