openLRSng/openLRSng #235: Project extension - STM32, Lora, Oled

danielsmd, 2020-11-23:
Hello.
I would like to extend the project with a STM32 based version. With my limited software skills, I manged to get running a demonstrator, with minimal functionality
[image]
I'm using STM32F103 + SX1278 RF module, and I have managed to get 98%-99% Mavlink telemetry @56k baud rate (57k air data rate), and FSK modulation. My initial idea was to swap on-the-go to Lora when the Link quality drops (and prioritize messages).
My intention is to configure everything from the TX, using the joystick (or the Bluetooth communication)
[image]

Since I'm a hardware engineer, I'm looking for software collaborators with spare time.... or at least to review/help me on the software side

mikeller, 2020-11-23:
@danielsmd: And this is compatible and interoperable with the existing openLRSng hardware? Or what else is it that makes it 'openLRSng'?

danielsmd, 2020-11-24:
well.. it will open source, and long range. It can be compatible (RF module changed ). This would be a extension/fork. I'm hoping people that use the system, are still interested in openLRSng to continue and develop.

mikeller, 2020-11-25:
@danielsmd: Not sure if re-using the protocol and logic that was developed to work on an 8 bit MCU with a relatively low-bandwidth 433 MHz RF chipset on a 32 bit MCU with much more capabilities and with a much higher bandwidth RF chipset is the correct approach.
If you are using modern hardware I'd rather start with a clean slate and implement a modern protocol that is not held back by compatibility with legacy hardware / software.

Or have a look at other projects that are out there and run on hardware similar to what you have in mind, and consider using these on your hardware: https://github.com/AlessandroAU/ExpressLRS